digistump/DigistumpArduino #167: link not working http://digistump.com/package_digistump_index.json

Murugesh-Hobbyist, 2024-04-25:
unable to add board in arduino IDE
http://digistump.com/package_digistump_index.json
not working.

vafaronaghi, 2024-05-04:
Try this one:

https://raw.githubusercontent.com/digistump/arduino-boards-index/master/package_digistump_index.json

Murugesh-Hobbyist, 2024-06-02:
> Try this one:
> 
> https://raw.githubusercontent.com/digistump/arduino-boards-index/master/package_digistump_index.json

Not working.
So Digistump is down?

pfeerick, 2024-06-02:
You should be able to use that url since all the resources it links are on github, not the digistump website. Otherwise. https://github.com/digistump/DigistumpArduino/issues/169#issuecomment-2092778200 for other BSP cores.

pfeerick, 2024-06-02:
I just loaded up a new copy the Arduino IDE, added https://raw.githubusercontent.com/digistump/arduino-boards-index/master/package_digistump_index.json via the Preferences dialog, then went to tools => board manager and was able to install the digistump avr boards package, and it is able to compile a blank sketch without any errors.

alanoo079, 2024-12-30:
> I just loaded up a new copy the Arduino IDE, added `https://raw.githubusercontent.com/digistump/arduino-boards-index/master/package_digistump_index.json` via the File => Preferences dialog, then went to Tools => Boards Manager and was able to install the digistump avr boards package, and it is able to compile a blank sketch without any errors. Make sure you removed the `http://digistump.com/package_digistump_index.json` one if you still have that there.

Yeah it works but how do i use libraries like DigiKeyboard.h? These dont work with it.

tai-yin, 2025-02-09:
Arduino returns error on https://raw.githubusercontent.com/digistump/arduino-boards-index/master/package_digistump_index.json.

The way I solved it is to simply download the json, fire up a simple backend(fastAPI in my case), host the file then give the endpoint to IDE like http://127.0.0.1:8000/package_digistump_index.json.

Example:

`from fastapi import FastAPI
from fastapi.responses import FileResponse

app = FastAPI()

@app.get("/package_digistump_index.json")
async def get_package_digistump_index():
    # Return the file as a response
    return FileResponse("package_digistump_index.json")`

MohamedAlwahashi, 2025-08-17:
1. The solution begins with defining the driver for the part for Windows.
[Click here](https://forum.arduino.cc/uploads/short-url/cMw5fnHVHxqVfOCBJE1XT6CYvcb.zip)

2. Then change the Arduino IDE program and delete it permanently. Delete all libraries and files previously added to the AppData folder in the "For All Users" section. Download the  [Arduino IDE v1.8.x version for Windows 7](https://www.arduino.cc/en/software/#ide).

> Because without this step, I was unable to add the boards to the program.

3. Then, before adding any files, add this file that our colleague added earlier:
https://raw.githubusercontent.com/digistump/arduino-boards-index/master/package_digistump_index.json
4. Then go to the board installation from the boards manager. You will find it in the list. Install it. Then go to the boards and you'll find the board added. Upload the code and make sure the board hasn't booted, as the boot time is very short. So, press the upload button, then connect the board, and the program will upload.
**If you want to change the code, make sure to press the upload button and then connect the board quickly, otherwise the program won't upload.**