digistump/DigistumpArduino #171: http://digistump.com down D:

Kidd3n, 2024-05-31:
[image]
Any other alternative to install the drivers to use the digispark

pfeerick, 2024-05-31:
Did you try searching the issues first?

https://github.com/digistump/DigistumpArduino/issues/167

Kidd3n, 2024-05-31:
thanks man :D

pfeerick, 2024-07-03:
> Did you try searching the issues first? #167
>
> Also, check out the BSPs mentioned here
> as this BSP is effectively discontinued since it has not had any updates or
> changes in seven years now.
>
> Pardon but whats a a "BSP"?

Board Support Package

arekm, 2024-07-14:
Use
https://raw.githubusercontent.com/digistump/arduino-boards-index/master/package_digistump_index.json