digistump/DigistumpArduino #173: Isn't working on Mac

the-trumpeter, 2024-09-28:
I've downloaded the `Digistump_Drivers.zip file`, and I've decompressed it into a folder. Inside the folder is a mess of files that loosely resembles an app. When I try to run the `launcher.exe` file, a notification opens saying that _You can’t open the application “launcher.exe” because Microsoft Windows applications are not supported on macOS._ When I paste the link to the file into Arduino IDE, I cannot find _Digistump AVR Boards_ in the Boards Manager.

Is there supposed to be Mac support? If so, where do I find the necessary files?

pfeerick, 2024-09-29:
Are you following the install instructions at https://github.com/digistump/DigistumpArduino ? As 

> http://digistump.com/package_digistump_index.json

is the install URL, but because the website has been down for sime time, now, the install URL for the board manager to use now is

> https://raw.githubusercontent.com/digistump/arduino-boards-index/master/package_digistump_index.json

But keep in mind this is still an old support package, i.e. pre Apple M1 silicon, so may not work for newer machines.

the-trumpeter, 2024-10-13:
yea i am on an m1 so hmmmm i might give it a try, if i do I'll put up my findings somewhere