digistump/DigistumpArduino #177: Issue with Digistump Post Install Script Blocked by Arduino IDE (post_install.bat not executed)

umutkayash, 2025-04-20:
I am currently using Arduino IDE version 1.8.19 (Windows Store version 1.8.57.0) on Windows 10, and I encountered a persistent warning message when trying to install and use the Digistump USB board support package.

After successfully adding the Digistump board via the Board Manager, the IDE displays the following warning message in orange at the bottom:

Warning: non trusted contribution, skipping script execution (C:\Users\Umut Kaya\Documents\ArduinoData\packages\digistump\tools\micronucleus\2.0a4\post_install.bat)
It appears that the post_install.bat script, which is essential for completing the Digistump setup (like driver installation), is not being executed due to being flagged as a "non trusted contribution." As a result, the board is not working properly and I'm unable to upload any sketches.

I would like to request clarification or a fix for this issue. Is there a way to manually allow trusted execution for this script? Or is this a bug in the Windows Store version of the Arduino IDE?

Steps to reproduce:

Install Arduino IDE from Microsoft Store.

Add Digistump AVR Boards from Board Manager (URL: http://digistump.com/package_digistump_index.json).

Observe the warning about skipped script execution.

System Info:

OS: Windows 10

Arduino IDE: 1.8.19 (Windows Store version 1.8.57.0)

Board: Digispark USB

Attached is a screenshot of the error message:
[file: reportjpg.png]

I appreciate any help or workaround you can offer. Thank you in advance for your support.

Best regards,
Umut Kaya

pfeerick, 2025-04-21:
Given the last change to this repository was (at time of writing) eight years ago... I would not hold my breath for a "fix"

[image]

Have a look at https://github.com/ArminJo/DigistumpArduino if you want an more updated version of the board support package, or  for still actively maintained BSP that supports those MCUs, https://github.com/SpenceKonde/ATTinyCore

per1234, 2025-04-23:
> Is there a way to manually allow trusted execution for this script?

Not execution via Arduino IDE 1.x. You can execute the batch file manually though.

>is this a bug in the Windows Store version of the Arduino IDE?

I would not consider it a bug because the developers intentionally designed Arduino IDE 1.x to skip the execution of [post-install scripts](https://arduino.github.io/arduino-cli/latest/platform-specification/#post-install-script) from 3rd party boards platforms (it only executes the scripts from the platforms maintained by the Arduino company.

However, after some consideration, it was determined that there is no basis for imposing such a restriction on 3rd party platforms (if the platform author wants to perform a malicious action, they can always use the commands that that Arduino IDE executes when you compile or upload a sketch while a board of that platform is selected). So this restriction was removed: https://github.com/arduino/arduino-cli/pull/911. However, that change was made after the time of the obsolete Arduino IDE 1.8.19 release you are using, so this is why you still encounter it.

This means that if you want Arduino IDE to execute the post-install scripts of 3rd party platforms, you only need to join me in the year 2025 and upgrade to Arduino IDE 2.x.