digistump/DigistumpArduino #179: digistump.com dead can we get a json that points here?

CTI-Tim, 2025-06-16:
digistump.com is dead-dead   
This means this link http://digistump.com/package_digistump_index.json   is also dead.
Can we please get a link and json for boards manager HERE so we can still use this?

KDMcMullan, 2025-06-26:
http://drazzy.com/package_drazzy.com_index.json
I THINK that includes it all.

pratik-londhe4, 2025-10-31:
@KDMcMullan , when I am using the drazzy one getting this error : 

```Failed to install platform: 'ATTinyCore:1.5.2'. 2 UNKNOWN: Get "https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86_64-apple-darwin.tar.bz2": tls: failed to verify certificate: x509: certificate has expired or is not yet valid: “azduino.com” certificate is expired```

it seems the site does not have valid certificate